arduino/arduino-ide #2622: The new IDE 2.0 editor no longer supports the CTL+"/" macro to comment out code block entries/exits?

rtek1000, 2025-02-05:
### Describe the problem

Hello,

I was using IDE 2.3.4 on Linux and everything was fine, but when I try to use it on Windows 11, I can't get the CTRL+"/" shortcut to work to comment out lines. It's CTRL+";".

Even in the advanced menu it was not possible to change to CTRL+"/". It's CTRL+";".

Apparently it has happened to other people before:
https://forum.arduino.cc/t/ide-editor-no-longer-supports-block-comment/698546

It shows the default as CTRL+";", but when I click edit, it says CTRL+"/", very confusing.

[image]

How do I leave CTRL+"/"?

### To reproduce

-

### Expected behavior

-

### Arduino IDE version

2.3.4

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

daxliniere, 2025-02-06:
Funny story.. ;)
Did you try both forward slash keys on your keyboard?

[image]

https://github.com/arduino/arduino-ide/issues/2621

rtek1000, 2025-02-06:
> Funny story.. ;) Did you try both forward slash keys on your keyboard?
> 
> [image]
> 
> [#2621](https://github.com/arduino/arduino-ide/issues/2621)

What are the two keys?

The first key "/" works on Linux, but not on Windows.

The second one is the key next to the number 0, which needs to be pressed with the FN key?

daxliniere, 2025-02-08:
> which needs to be pressed with the FN key?

Depends heavily on your keyboard layout and the default function of your keyboard keys. Easiest way to know is to click in the text editor area of the IDE ('the main window') and make the key type a forward slash. Then, whatever key combo got you there, try that in the shortcut editor.

daxliniere, 2025-02-08:
Hopefully we can have a reply from of the devs next week.

rtek1000, 2025-02-09:
If I use Windows 7 x64 in a virtual machine, installed on the Linux host, the problem also occurs. It seems that the problem is on the notebook. Although IDE 1.8.19 works normally, both on Linux and Windows 11.

It's difficult to understand that in a text editor the keys work correctly, but not in shortcuts. And if I try to leave CTRL+"\", then it appears as CTRL+"]" in the settings, and none of these combinations work.

daxliniere, 2025-02-09:
> It's difficult to understand that in a text editor the keys work correctly, but not in shortcuts.

Yes, this is the whole point of why this is a bug and I reported it. :)