arduino/arduino-ide #2634: Serial monitor is a memory+CPU hog

daxliniere, 2025-02-16:
### Describe the problem

I have been working on a program and, so I can tell where in the various loops I am, I have it serial print a single character at the start of each loop.

The trouble is that after a minute, the entire IDE starts to run very slowly. After a couple more minutes, it is unable to summon the CPU power  required to compile code, let alone attempt to upload it. Even scrolling your code becomes impossible.
Unplug your USB cable from Arduino, and after a few seconds, everything becomes springy and light again.

### To reproduce

Run this:
```
void setup() {
  Serial.begin(115200);
}

void loop() {
Serial.print("A");
loop2();
}

void loop2() {
Serial.print("B");
}
```

### Expected behavior

Processing priority of the serial monitor should be lower than the text editor, compiler, and probably some other things, too.

### Arduino IDE version

2.3.4

### Operating system

Windows

### Operating system version

10 Pro

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

daxliniere, 2025-02-16:
Additional info! I have just discovered that if you press the Clear Output button, most of the speed is restored... until it fills up again.

I have also decreased my baud rate, but the issue still stands.

per1234, 2025-02-16:
Hi @daxliniere . Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/934.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

---

The workaround is to adjust your sketch code so that it will produce line breaks as appropriate to avoid the output of excessively long lines. In your sketch, this would be accomplished by simply changing the `Serial.print` calls to `Serial.println`:

```cpp
void setup() {
  Serial.begin(115200);
}

void loop() {
Serial.println("A");
loop2();
}

void loop2() {
Serial.println("B");
}
```

closed 2025-02-16 by per1234

daxliniere, 2025-02-16:
Thanks for the professionalism of your replies to each of my recent bug reports, Per.
It seems like I keep finding known issues that have been known for a long time, some fore many years. Is it fair to say that the Arduino IDE is a dead project? Seems like tiny easy-fix things are not being fixed, and some big things are also not being fixed. Prehaps there is not enough money in Arduino LLC to pay for the correct amount of developers' time to fix these issues in a timely manner?