arduino/arduino-ide #2650: Arduino IDE 2.3.4 on Apple M4 Pro not Working

marcellofiore, 2025-03-02:
### Describe the problem

Unfortunately after trying several times, I have to find that Arduino ide remains in loading state and is not usable. It remains in loading state with the animated zoom in and zoom out logo.

### To reproduce

I download and install Arduino IDE from => https://www.arduino.cc/en/software the “macOS” version
Apple Silicon, 11: “Big Sur” or newer, 64 bit”

[image]

### Expected behavior

opening

### Arduino IDE version

2.3.4

### Operating system

macOS

### Operating system version

Sequoia 15.3.1

### Additional context

Apple M4 Pro (Apple Silicon)

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-03-02:
Hi @marcellofiore. I'm going to ask you to provide some additional information that might help us to identify the problem.

---

❗ This procedure is not intended to solve the problem. The purpose is to gather more information.

---

Please do this:

1. Close all Arduino IDE windows if the IDE is already running.
1. [Open a "**Terminal**" window](https://support.apple.com/guide/terminal/apd5265185d-f365-44cb-8b09-71a064a42125/mac#trml663c6997).
1. Type the following command:
   ```text
   "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
   ```
1. Press the <kbd>**Enter**</kbd> key.
   Debug output should now be printed to the **Terminal** window as Arduino IDE attempts to start.
1. Wait until you are sure the IDE startup has gone into the hang state
   (as indicated by no additional output being printed to the **Terminal** window).
1. Switch back to the **Terminal** window if the Arduino IDE window took the focus.
1. Press the <kbd>**Command**</kbd>+<kbd>**A**</kbd> keyboard shortcut.
   This will select all the text in the **Terminal** window.
1. Press the <kbd>**Command**</kbd>+<kbd>**C**</kbd> keyboard shortcut.
   This will copy the contents of the **Terminal** window to the clipboard.
1. Paste (<kbd>**Command**</kbd>+<kbd>**V**</kbd>) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

marcellofiore, 2025-03-02:
Last login: Sun Mar  2 18:09:21 on ttys000
marcellofiore@Mac ~ % "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE" 
Arduino IDE 2.3.4
Checking for frontend application configuration customizations. Module path: /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/electron-main.js, destination 'package.json': /Applications/Arduino IDE.app/Contents/Resources/app/package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.4" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "1.1.1" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2024-12-03T09:58:49.900Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{}},"defaultLocale":"","validatePreferencesSchema":false,"preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.4","cliVersion":"1.1.1","buildDate":"2024-12-03T09:58:49.900Z"}
Starting backend process. PID: 43993
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///Users/marcellofiore/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2025-03-02T17:13:57.736Z root INFO Backend Object.initialize: 1.4 ms [Finished 0.196 s after backend start]
2025-03-02T17:13:57.737Z root INFO Backend Object.configure: 1.1 ms [Finished 0.196 s after backend start]
2025-03-02T17:13:57.737Z root INFO Backend s.onStart: 0.5 ms [Finished 0.196 s after backend start]
2025-03-02T17:13:57.737Z root INFO Backend s.configure: 0.1 ms [Finished 0.204 s after backend start]
2025-03-02T17:13:57.737Z root INFO Backend d.onStart: 0.1 ms [Finished 0.204 s after backend start]
2025-03-02T17:13:57.738Z root INFO Backend d.configure: 0.0 ms [Finished 0.205 s after backend start]
2025-03-02T17:13:57.738Z root INFO Backend d.initialize: 8.4 ms [Finished 0.205 s after backend start]
2025-03-02T17:13:57.738Z root INFO Backend c.onStart: 0.1 ms [Finished 0.205 s after backend start]
2025-03-02T17:13:57.738Z root INFO Backend a.configure: 0.4 ms [Finished 0.205 s after backend start]
2025-03-02T17:13:57.739Z config INFO >>> Initializing CLI configuration...
2025-03-02T17:13:57.739Z root INFO Backend c.initialize: 0.2 ms [Finished 0.205 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend x.onStart: 0.5 ms [Finished 0.206 s after backend start]
2025-03-02T17:13:57.739Z config INFO Loading CLI configuration from /Users/marcellofiore/.arduinoIDE/arduino-cli.yaml...
2025-03-02T17:13:57.739Z daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/resources/arduino-cli...
2025-03-02T17:13:57.739Z root INFO Backend a.initialize: 0.2 ms [Finished 0.206 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend w.onStart: 0.4 ms [Finished 0.206 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend r.configure: 0.7 ms [Finished 0.206 s after backend start]
2025-03-02T17:13:57.739Z discovery-log INFO start
2025-03-02T17:13:57.739Z discovery-log INFO start new deferred
2025-03-02T17:13:57.739Z root INFO Backend d.initialize: 0.1 ms [Finished 0.206 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend y.onStart: 0.6 ms [Finished 0.207 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend r.configure: 0.5 ms [Finished 0.207 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend a.onStart: 0.2 ms [Finished 0.207 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend h.configure: 0.1 ms [Finished 0.207 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend s.configure: 0.0 ms [Finished 0.207 s after backend start]
2025-03-02T17:13:57.739Z root INFO Backend l.configure: 0.0 ms [Finished 0.207 s after backend start]
2025-03-02T17:13:57.740Z root INFO Theia app listening on http://127.0.0.1:53375.
2025-03-02T17:13:57.740Z root INFO Finished starting backend application: 0.8 ms [Finished 0.208 s after backend start]
2025-03-02T17:13:57.742Z root WARN The local plugin referenced by local-dir:/Users/marcellofiore/.arduinoIDE/plugins does not exist.
2025-03-02T17:13:57.742Z root WARN The local plugin referenced by local-dir:/Users/marcellofiore/.arduinoIDE/extensions does not exist.
2025-03-02T17:13:57.752Z root INFO Resolve plugins list: 12.8 ms [Finished 0.220 s after backend start]
2025-03-02T17:13:57.798Z daemon INFO time="2025-03-02T18:13:57+01:00" level=info msg="arduino-cli version 1.1.1"
2025-03-02T17:13:57.798Z daemon INFO Error: path non valido per la creazione della cartella di configurazione: errore /Users/marcellofiore/Library/Arduino15: mkdir /Users/marcellofiore/Library/Arduino15: permission denied
**2025-03-02T17:13:57.798Z daemon ERROR Error: Error: path non valido per la creazione della cartella di configurazione: errore /Users/marcellofiore/Library/Arduino15: mkdir /Users/marcellofiore/Library/Arduino15: per [... cut ...]
2025-03-02T17:13:57.834Z daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/resources/arduino-cli...
2025-03-02T17:13:57.834Z daemon INFO Daemon exited with exit code: 10.
2025-03-02T17:13:57.853Z daemon INFO time="2025-03-02T18:13:57+01:00" level=info msg="arduino-cli version 1.1.1"
2025-03-02T17:13:57.853Z daemon INFO Error: path non valido per la creazione della cartella di configurazione: errore /Users/marcellofiore/Library/Arduino15: mkdir /Users/marcellofiore/Library/Arduino15: permission denied.....

marcellofiore, 2025-03-02:
@per1234 I think it lacks write permissions to be able to create the folders needed to boot up!

giacomocusinato, 2025-03-03:
Seems Arduino CLI does not have the necessary permissions to create the default directories for .

Can you try typing `ls -ld ~/Library` in the terminal and post the result?
This can help verify any unusual permission applied to the library folder.

marcellofiore, 2025-03-03:
> Seems Arduino CLI does not have the necessary permissions to create the default directories for its configuration.
> 
> [@marcellofiore](https://github.com/marcellofiore) can you try typing `ls -ld ~/Library` in the terminal and post the result? This can help verify any unusual permission applied to the library folder.

drwxrwxr-x@ 91 root  wheel  2912 Mar  2 18:22 /Users/marcellofiore/Library

distancerunner, 2025-03-24:
Today, on my TuxOS/KDE machine, the ./arduino-ide_2.3.4_Linux_64bit.AppImage also not starting. 

So I tried to start it in terminal. And it is a permission error:

> 200391:0324/090947.898939:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_arduinNmP2VS/chrome-sandbox is owned by root and has mode 4755.
Trace/Breakpoint ausgelöst

starting with is working fine:
> ./arduino-ide_2.3.4_Linux_64bit.AppImage --no-sandbox

grimace, 2025-10-01:
I have this same problem with a fresh download today.  None of these suggestions worked.  What is the next step?
I'm running 
    Apple M3 Max
    Sequoia 15.6.1 (24G90)
App starts, puts up message to say ( application downloaded from internet do you want to open --> yes )
spinning beach ball for 5-10 seconds, then nothing
Apple-Tab shows an ArduinoIDE logo but no window ever shows up, nor is there an entry in the App bar at the top ( next to the Apple logo ) ever showing up.  This is with the production version and with the latest nightly:
arduino-ide_nightly-20251001_macOS_arm64.dmg

justinsousa, 2026-01-05:
my issue here seems to be that it's trying to mkdir in `/Users/$(whoami)_1/`  instead of just `/Users/$(whoami)/` . I just downloaded the .dmg from the site to install. Not sure where the _1 would be coming from