arduino/arduino-ide #2663: Can't figure out how to select each of two Teensy boards

billtubbs, 2025-03-16:
### Describe the problem

I have two Teensy boards attached to my MacBook using a USB hub.  They both show up in Arduino IDE here:

[image]

and also in the ports menu:

[image]

I'm not sure whether I am supposed to select the Teensy I want from the ports menu or the boards drop-down (or both) but whatever I select it doesn't seem to matter – the app seems to upload to the second Teensy regardless, after the following message in the compiler window:

```
Found 2 Teensy boards, but using auto-search to find board for upload.
Please use Tools > Ports(Teensy) to select the specific board.
```

### To reproduce

1. Connect two Teensy 3.2 boards via a USB hub
2. Load a script
3. Select the first board in Tools -> ports and click the 'Upload' button
4. Observe which board received the compiled program
5. Select the second board in Tools -> ports and click the 'Upload' button
6. Observe which board received the compiled program.

### Expected behavior

Script is compiled and uploaded to the board selected in ports.
No "Found 2 Teensy boards" message is displayed

### Arduino IDE version

2.3.4

### Operating system

macOS

### Operating system version

10.15.7

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-03-16:
Hi @billtubbs.

>Teensy Loader: 1.58.2

I think this is the problem. The problem you are reporting is a bug in older versions of the Teensy tools:

https://forum.pjrc.com/index.php?threads/arduino-ide-2-0-0-selection-usb-port-when-using-2-teensy4-0-parallel.71112/

The bug was fixed in version 1.59 of the tools:

https://forum.pjrc.com/index.php?threads/ide-2-and-multiple-t_4-1s-and-a-single-t_4-0-only-sees-the-t_4-1s.73125/post-338252

> The issue reported here is fixed in [1.59-beta5](https://forum.pjrc.com/index.php?threads/teensyduino-1-59-beta-5.74262/).

So you only need to use the [Arduino IDE Boards Manager](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-board-manager) to update your Teensy boards platform to the latest version:

https://www.pjrc.com/arduino-ide-2-0-0-teensy-support/

If you would like assistance with updating the platform, you can post on Arduino Forum:

https://forum.arduino.cc/c/development-tools/ide-2-x/93

or the Teensy Forum:

https://forum.pjrc.com/

closed 2025-03-16 by per1234

billtubbs, 2025-03-16:
Thanks.  Didn't realize it was a problem with the Teensy loader.  I will upgrade as soon as they fix another bug.  Thanks for your help.