arduino/arduino-ide #2667: translated strings not sync with transifex at least 1 year ?!

coby2023t, 2025-03-21:
### Describe the problem

many updated or new strings not update yet.

### To reproduce

take a look

### Expected behavior

check the build procedure out and sync the po files everytime.

### Arduino IDE version

2.x

### Operating system

Linux

### Operating system version

linux

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-03-21:
Thanks for your report @coby2023t. We typically sync with Transifex before each production release. You can see that was done prior to the latest 2.3.4 release:

https://github.com/arduino/arduino-ide/pull/2523

I have verified that all changes to the translations from that pull request are in version 2.3.4 of Arduino IDE.

It is true that any translations made since 2024-12-01 are not synced, but those would only be available in the [nightly build](https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md#testing-nightly-build) of Arduino IDE anyway. For this reason, I don't think that the lack of syncs in the interval between releases is very impactful to the user base (since most users reasonably use the release version of Arduino IDE).

So my findings don't match your "at least 1 year" claim. Please provide a detailed description of what led you to this conclusion in a comment here on the issue thread and I'll investigate further.

coby2023t, 2025-04-01:
that make sense, my bad, didn't know the the linux dist do not sync with nightly built one.

closed 2025-04-01 by coby2023t