arduino/arduino-ide #2678: Cannot open multiple sketches at once

Gartosz, 2025-04-02:
### Describe the problem

When I have one sketch opened opening another (different) one with double clicking on .ino file will result in a new window with the same sketch. When I use open option in the second window and choose the sketch I wanted to open it creates third window with correct file.

### To reproduce

1. Open one sketch.
2. Open different sketch by double clicking on an ino file.
3. Both windows show the first one. It goes forever.

### Expected behavior

I should have multiple sketches in multiple windows without need to use open option in the program.

### Arduino IDE version

2.3.5

### Operating system

Windows

### Operating system version

11

### Additional context

That behaviour started occuring after the last update.

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

Gartosz, 2025-04-02:
Also during compile in one window I cannot use top menu in the second one. It closes itself whenever another row is printed in the output. Seems like it impacts other windows too.

nickthesafe, 2025-04-03:
Confirmed here, the issue is as Gartosz states; have one sketch open and attempt to open further instances of Arduino IDE containing alternate sketches. The program simply duplicates the previously opened instance into the new window and not the file as requested.

kwagnitz, 2025-04-05:
Ditto on Ubuntu Linux 24.04.
This has been a problem for a very long time.  Makes the IDE confusing to use. 
I don't know why it is so hard to correct.

Gartosz, 2025-04-05:
> Ditto on Ubuntu Linux 24.04. This has been a problem for a very long time. Makes the IDE confusing to use. I don't know why it is so hard to correct.

Then why haven't you reported that when the issue firstly occured?

closed 2025-04-05 by giacomocusinato

kwagnitz, 2025-04-06:
Just downloaded _arduino-ide_nightly-20250406_Linux_64bit.AppImage_.
Problem still exists there.  If one instance of the IDE has a file open, clicking on a second file, brings up a second instance showing the first file, not the second one clicked on.
(Ubuntu 24.04)

kwagnitz, 2025-04-06:
I must apologise, @per1234.

I was accidentally using an older appimage.
Using 
_Version: 2.3.6-nightly-20250406
Date: 2025-04-06T03:21:50.829Z
CLI Version: 1.2.0_
I can confirm that the correct file gets opened.
Tried it with 3 files simultaneously.  3 instances of the IDE opened, each with the different .ino that I'd clicked on.
Very glad it is fixed.  It sure confused me each time I forgot that the behaviour was to show the same file in each instance.

KISPE-Harvey, 2025-04-25:
@per1234 
After upgrading to 2.3.6 from the Arduino software page, I restarted and still had the same issue and reading your previous post was able to get the version etc.

Version: 2.3.6
Date: 2025-04-09T11:26:55.498Z
CLI Version: 1.2.0

However i did see you mention about the nightly builds, installed the latest from the website and still have the same issue

Version: 2.3.7-nightly-20250423
Date: 2025-04-23T03:25:41.256Z
CLI Version: 1.2.0

Harveyn4444, 2025-04-25:
@per1234 
After upgrading to 2.3.6 from the Arduino software page, I restarted and still had the same issue and reading your previous post was able to get the version etc.

Version: 2.3.6
Date: 2025-04-09T11:26:55.498Z
CLI Version: 1.2.0

However i did see you mention about the nightly builds, installed the latest from the website and still have the same issue

Version: 2.3.7-nightly-20250423
Date: 2025-04-23T03:25:41.256Z
CLI Version: 1.2.0