arduino/arduino-ide #2683: opens wrong .ino file

smdg49, 2025-04-03:
### Describe the problem

if you open fileA.ino through windows explorer and while fileA is open, you open fileB.ino. Arduino IDE will just open another fileA window and not even open fileB.

### To reproduce

Go to a sketch folder. Open any sketch. Then go to a different Sketch folder and open another sketch. Arduino IDE will just reopen first sketch.

### Expected behavior

Arduino IDE should open the sketch you double clicked.

### Arduino IDE version

2.3.5

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-04-03:
Hi @smdg49. This is a duplicate of https://github.com/arduino/arduino-ide/issues/2678.

In the future, please take the time to search the issue tracker for existing issues before submitting one here. You claimed to have done that:

> - [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)

However, it is very clear you did not because the existing issue report was at the very top of the issues list!

closed 2025-04-03 by per1234

smdg49, 2025-04-04:
That post said something about command line argument. Was not sure it was
the same issue.

smdg49, 2025-04-04:
> Hi [@smdg49](https://github.com/smdg49). This is a duplicate of [#2678](https://github.com/arduino/arduino-ide/issues/2678).
>
> In the future, please take the time to search the issue tracker for [...]

That post stated "command line argument while IDE already running" had me thinking that it was a different issue