arduino/arduino-ide #2685: Compilation errors after upgrade from 2.3.4 to 2.3.5

KenkiCK, 2025-04-04:
### Describe the problem

After Upgrade the Arduino IDE I get compilation errors for the Heltec_ESP32_Dev-Boards library. Before  the upgrade it works without problems.

### To reproduce

Compile a heltec lora 32 (V3) board code.  

```
#include <WiFi.h>
#include <PubSubClient.h>
#include <HT_SSD1306Wire.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Expected behavior

I get these errors now 

> d:\Chris\Dokumente\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\sx126x.c: In function 'sx126xSleep':
> d:\Chris\Dokumente\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\sx126x.c:247:5: error: implicit declaration of function 'delay'; did you mean 'Delay'? [-Wimplicit-function-declaration]
>   247 |     delay( 2 );
>       |     ^~~~~
>       |     Delay
> d:\Chris\Dokumente\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\sx126x.c: In function 'SX126xGetPacketStatus':
> d:\Chris\Dokumente\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\sx126x.c:731:13: error: implicit declaration of function 'memset' [-Wimplicit-function-declaration]
>   731 |             memset( pktStatus, 0, sizeof( PacketStatus_t ) );
>       |             ^~~~~~
> d:\Chris\Dokumente\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\sx126x.c:9:1: note: 'memset' is defined in header '<string.h>'; this is probably fixable by adding '#include <string.h>'
>     8 | #include "esp_sleep.h"
>   +++ |+#include <string.h>
>     9 | /*!
> d:\Chris\Dokumente\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\sx1262-board.c: In function 'SX126xWaitOnBusy':
> d:\Chris\Dokumente\Arduino\libraries\Heltec_ESP32_Dev-Boards\src\driver\sx1262-board.c:82:17: error: implicit declaration of function 'lora_printf'; did you mean 'log_printf'? [-Wimplicit-function-declaration]
>    82 |                 lora_printf("spi timeout\r\n");
>       |                 ^~~~~~~~~~~
>       |                 log_printf
> exit status 1
> 
> Compilation error: exit status 1

### Arduino IDE version

2.3.5

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

ClaudeBoemare, 2025-04-05:
I had  a similar problem, I solved by downgrade the expressif board file from 3.2.0 to 3.1.3

KenkiCK, 2025-04-05:
I use the Heltec ESP32 Series develop environment - actual version 3.0.2 - the preverious version 3.0.0 is not available anymore but in the past it works with 3.0.2.

A board which you mentioned I can't find.

closed 2025-04-05 by KenkiCK

jgrowdon, 2025-04-05:
Hey KenkiCK - I had the same problem.  I decoded the error messages and fixed the problem by editing two of the include files:

1. sx1262-boards.c - at the top of this file, change the statement "lora_printf" to "log_printf"
2. radio.c - you have to add the standard Arduino include file:  #include "Arduino.h"

When I made those two changes in the src/driver files, it fixed the compile problems.

John G.

MSchmidke, 2025-04-19:
Came along here having exactly the same issue. May I ask why it has been closed?