arduino/arduino-ide #2698: Arduino IDE V2.3.5 Compilation Process Frequently Steals Focus, Causing Keyboard Input Interruption (Pop-up Windows Always on Top)

YFROBOTZL, 2025-04-09:
### Describe the problem

During code compilation (Verify/Upload), Arduino IDE 2.3.5 forcibly captures mouse focus and displays pop-up windows (e.g., log panels or error prompts) that remain on top, preventing users from using the keyboard for other operations (e.g., switching windows or editing code). This issue is particularly noticeable in complex projects or during prolonged compilation, severely impacting development efficiency.​​
​​Examples​​:

The IDE main window or pop-ups stay on top during compilation, blocking input to other applications.

### To reproduce

Windows 10 64BIT

complie and upload  show verbose output during
[image]

Trigger Compilation: Click "Verify" or "Upload".

### Expected behavior

Feature Suggestion: Propose adding a "Silent Compilation Mode" to disable non-critical pop-ups

### Arduino IDE version

2.3.5

### Operating system

Windows

### Operating system version

Windows 10 22H2

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

YFROBOTZL, 2025-04-09:
I have reverted to version 2.3.4 and this situation did not occur.

per1234, 2025-04-09:
Duplicate of https://github.com/arduino/arduino-ide/issues/2679, which has already been fixed by https://github.com/arduino/arduino-ide/pull/2681.

>- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)

@YFROBOTZL you checked that box to attest that you had tested with the [nightly build](https://www.arduino.cc/en/software#nightly-builds) of Arduino IDE, but clearly you didn't actually do that. In the future, please do test with the nightly build of Arduino IDE before submitting an issue to avoid creating duplicates.

closed 2025-04-09 by per1234

YFROBOTZL, 2025-04-09:
OK