arduino/arduino-ide #2700: ファイルを開いた状態で新しく違うプログラムを開いても同じプログラムが実行される

miki-wmhne, 2025-04-09:
### Describe the problem

プログラムを 開いた状態で 新しく違うプログラムを開いても 同じプログラムが表示されてしまいます
Ver.2.3.4は問題無く2個のプログラムが表示出来ています

### To reproduce

Arduino ホルダー内のプログラムをＩＤＥで開きます　次に開いた状態で 新しく違うプログラムを開きます　しかし後から開いたプログラムは
表示が出来ず 同じプログラムが2個 表示になります

### Expected behavior

2個のプログラムを開いて編集が出来ない

### Arduino IDE version

Ver.2.3.5

### Operating system

Windows

### Operating system version

Windows11 24H2 New UpDate

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-04-09:
Duplicate of https://github.com/arduino/arduino-ide/issues/2678, which has already been fixed by https://github.com/arduino/arduino-ide/pull/2686.

>- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)

@miki-wmhne you checked that box to attest that you had tested with the [nightly build](https://www.arduino.cc/en/software#nightly-builds) of Arduino IDE, but clearly you didn't actually do that. In the future, please do test with the nightly build of Arduino IDE before submitting an issue to avoid creating duplicates.

closed 2025-04-09 by per1234

miki-wmhne, 2025-04-09:
arduino IDE  Ver.2.3.5 の直後の Arduino
IDEのナイトリービルドで確認したら同じ現象でした。今日、arduino-ide_nightly-20250409_Windows_64bitwをDownloadして確認した所
改善されてました。
正式にアップデートをお願いします。 それとファイルの保存が Saveになっています。