arduino/arduino-ide #2702: Ver.2.3.6でタスクメニューの一部が日本語対応になっていません

miki-wmhne, 2025-04-10:
### Describe the problem

ファイルを開いた所の保存が Sabeのまま表示　編集を開いた所の貼り付けがPasteのまま表示　
Ver2.3.4 では Sabeは同じでしたが　貼り付けは正常に表示出来ていました

### To reproduce

ソフトのバグなので再現は無くてもインストールすれば確認が出来ます

### Expected behavior

正常に日本語対応になります

### Arduino IDE version

Ver.2.3.6     ナイトリービルド20250409で確認

### Operating system

Windows

### Operating system version

Windows11 24H2

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

giacomocusinato, 2025-04-10:
Thanks @miki-wmhne, this seems to be a problem with all languages. 
I've added some details in english to better keep track of this.

miki-wmhne, 2025-04-10:
It is looking forward to the style corrected in an update next time.