arduino/arduino-ide #2712: Feature Request: Add support for Real Digital Blackboard board

Cool-Coder174, 2025-04-17:
### Describe the request

I’d love to see first‑class Arduino IDE support for the [Real Digital Blackboard](https://www.realdigital.org/hardware/blackboard) board used at Washington State University. That would include:

A Board Manager JSON package entry

Core libraries or BSP for the Zynq‑7000 PS (bare‑metal and/or Linux)

Upload tooling (e.g. OpenOCD, USB‑JTAG, Serial‑bootloader)

Example sketches and README

Educational use: The Blackboard is heavily used in embedded systems and digital design courses at Washington State University and elsewhere.

Zynq integration: Gives Arduino users hands‑on exposure to hybrid PS/PL platforms without leaving the IDE.

Community growth: Extends Arduino’s ecosystem into FPGA‑SoC territory and opens up new project possibilities.

**Hardware Details**
SoC: Xilinx Zynq XC7Z007S (Dual‑core ARM Cortex‑A9 + Artix‑7 FPGA)

I/O: PS‑side UART, I²C, SPI, GPIO (MIO, EMIO), two Pmod® headers, USB OTG/Host, SD card slot, on‑board sensors

Voltage levels: 1.8–3.3 V PS I/O, 3.3 V PL I/O

Docs & schematics: https://www.realdigital.org/hardware/blackboard

4. Proposed Implementation
Board Manager entry

Add a JSON manifest under https://downloads.arduino.cc/packages/real_digital_blackboard/package_blackboard_index.json

Point to a zipped core/BSP with hardware/blackboard/1.0.0 folder.

Core/BSP

Provide low‑level C headers for PS peripherals (UART, I²C, SPI, GPIO).

Optionally include a PetaLinux BSP or bare‑metal examples.

Upload tool support

Leverage OpenOCD flashing over JTAG or the PS serial‑bootloader for bare‑metal.

Expose a “Blackboard” programmer option in Tools → Programmer.

**Examples & docs**

A set of “Hello world” sketches: blink the on‑board LED, read I²C temp sensor, drive a Pmod.

**README** with wiring diagrams and step‑by‑step getting started.

**Alternatives Considered**
Generic “Zynq” board in Board Manager (too vague; users need the right pin mapping).

Manual toolchain setup outside the IDE (steep learning curve for students).

### Describe the current behavior

No current support for BlackBoard

### Arduino IDE version

 Arduino IDE 2.3.6

### Operating system

macOS

### Operating system version

MacOS 15.4

### Additional context

More info on blackboard: https://www.realdigital.org/hardware/blackboard

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

Cool-Coder174, 2025-04-17:
/assign

per1234, 2025-04-18:
Hi @Cool-Coder174. This issue tracker is only used to track bug reports and feature requests for the Arduino IDE 2.x application codebase hosted in this repository. Although your proposal is a good one, it is completely unrelated to this codebase, and so I will close this as off topic.

The project you propose would either be the responsibility of the board manufacturer, or of the Arduino community. The Arduino company's responsibility is to provide a framework by which support can be added to the official Arduino development tools for any development board and microcontroller. We have already provided that framework, which has been used by the Arduino community to provide support for a vast array of microcontrollers and 3rd party boards, in addition to Arduino's own work to provide support for the official boards.

You can learn how to create an Arduino boards platform for the "Real Digital Blackboard" from the documentation here:

https://arduino.github.io/arduino-cli/latest/platform-specification/

https://arduino.github.io/arduino-cli/latest/package_index_json-specification/

You can check to see what might have already been done in related areas by checking the lists of existing Arduino boards platforms here:

https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

and more here:

https://github.com/per1234/inoplatforms/blob/main/ino-hardware-package-list.tsv

If you would like to discuss the subject, request assistance with the project, or share your finished work, you are very welcome to post over on Arduino Forum:

https://forum.arduino.cc/

closed 2025-04-18 by per1234