arduino/arduino-ide #2715: Kernel Panic on macOS (Apple Silicon) when Flashing NRF52840 via Arduino-cli

rsispal, 2025-04-19:
### Describe the problem

Hi all, 

I'm experiencing sporadic kernel panics on my macOS (Apple Silicon) system when attempting to flash an NRF52840 board (Seeed Studio XIAO NRF52840) using the Arduino IDE, specifically when the flashing process is executed by arduino-cli. The panics consistently occur during the upload, leading to a system crash and restart.    

As it happens so forcefully and frequently (over 20 times these past couple of days) I have noticed that it happens at the following stage:

```sh
Performing 1200-bps touch reset on serial port /dev/cu.usbmodem101
```

I've tried changing ports, USB-C cables and even switched to using a USB-A to USB-C cable via a USB-C adapter (Apple official)

I've got the logs available and will upload one, anyone else experiencing this?

I've never had a crash with an ESP32 board, only the NRF52840 which makes me think there's a driver issue in relation to Nordic.

### To reproduce

Steps to Reproduce:

Connect an NRF52840 board to the macOS system (SEEED studio xiao NRF52840)
Open the Arduino IDE and configure it for the NRF52840 board.
Attempt to upload a sketch to the board. (Specify if it's a simple or complex sketch. If it's a specific sketch, include it or a link to it.)
When it gets to `Performing 1200-bps touch reset on serial port ` observe a kernel panic
The system will kernel panic during the upload process.

In my case there's a crash every 1 in 10 flashes (so far)

### Expected behavior

The sketch should upload to the NRF52840 board without causing a system-level kernel panic.

Actual Behavior:

The macOS system crashes and restarts with a kernel panic. The error is a Kernel data abort. 

### Arduino IDE version

Version: 2.3.6 Date: 2025-04-09T11:22:51.016Z CLI Version: 1.2.0

### Operating system

macOS

### Operating system version

Sequoia 15.4 (24E248)

### Additional context

The crash doesn't happen every time, but is sufficient enough to cause extreme frustration.

I note that the same crash happens via PlatformIO in the same 1200 baud reset step.

[panic-full-2025-04-19-230931.0002.txt](https://github.com/user-attachments/files/19823076/panic-full-2025-04-19-230931.0002.txt)

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

vdmprojects, 2025-04-29:
I can replicate the behavior. It happens frequently, every few uploads. 
It is really, really annoying lol.

demshy, 2025-10-08:
Hi, I know this is old and forgotten, but for anyone who has hopes that the new OS version would help: it doesnt.

Still happening on Tahoe 26.0.1 on my M1. 26.0.1 :(

elliotmatson, 2026-11-06:
I just started seeing what seems to be this issue yesterday on macOS 27.2. Seems to be a NULL dereference in com.apple.driver.usb.cdc. This probably isn't an Arduino issue, but it's probably worth keeping this open for reference (I'm seeing this with a python upload script that calls esptool, so a completely different chip and build environment)