arduino/arduino-ide #2725: save as behaivour changed

aalucia, 2025-04-30:
### Describe the problem

 in Arduino IDE 2.3.6, the behavior of "Save As" has changed compared to previous versions. Previously, the IDE would remember the last used directory and open the "Save As" dialog in that folder. However, in 2.3.6, it defaults to the Arduino default sketch directory (usually Documents/Arduino), regardless of where you last saved or opened a sketch.

### To reproduce

use "save as"

### Expected behavior

should go to the last used directory, but goes to default one instead

### Arduino IDE version

2.3.6

### Operating system

Linux

### Operating system version

Suse linux

### Additional context

until 2.3.4 did work correctly, note: cannot install lates nighly build

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

per1234, 2025-05-01:
Hi @aalucia. Thanks for your report. I just did some testing with Arduino IDE 2.3.6 on a Linux machine and was not able to reproduce the fault. For me, the behavior of which folder is selected in the "Save As" dialog by default is exactly the same in 2.3.6 as in 2.3.4.

It might be that my failure to reproduce the fault is because I am not using the correct procedure. Please provide detailed step by step instructions I can follow to reliably reproduce the fault on my computer.

aalucia, 2025-05-02:
Hi,
the steps are:
1) open a sketch from "recent ones"  (or choose a sketch form a directory)
2) save as -> saving dialog proposes the default directory and not the one from where the "recent" comes from

On version 2.3.4 instead a new windows pop up and is possible to choose from "bookmarked" directories.
System is SUSE linux:

Operating System: openSUSE Leap 15.6
KDE Plasma Version: 5.27.11
KDE Frameworks Version: 5.115.0
Qt Version: 5.15.12
Kernel Version: 6.4.0-150600.23.47-default (64-bit)
Graphics Platform: X11
Processors: 16 × AMD Ryzen 7 5800X3D 8-Core Processor
Memory: 62.7 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080 Ti/PCIe/SSE2
Manufacturer: Micro-Star International Co., Ltd.
Product Name: MS-7C37
System Version: 2.0
Alex.

per1234, 2025-06-03:
Thanks for the clarification. Unfortunately I am still unable to reproduce the fault.

I performed the following procedure with Arduino IDE 2.3.6 on a Ubuntu 24.04.2 machine:

1. Select **File > New Sketch** from the Arduino IDE menus.
   A new Arduino IDE window opens.
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch at `~/Arduino/foo/bar`.
1. Select **File > Close** from the Arduino IDE menus.
   The "bar" sketch window closes.
1. Select **File > Open Recent > bar** from the Arduino IDE menus.
   The "bar" sketch opens in a new Arduino IDE window.
1. Select **File > Save As...** from the Arduino IDE menus.

The "Save As" dialog opens with the `~/Arduino/foo` folder selected:

[image]

This seems to be the correct behavior, whereas if the fault had occured as you described I would have expected it to instead open with the `~/Arduino` folder selected (I have my "**Sketchbook folder**" preference set to the `~/Arduino` folder).

@aalucia if you perform the procedure I described above, do you get a different result than I did?

I did find some reports of the problem on Arduino Forum:

- https://forum.arduino.cc/t/laggy-arduino-ide-2-0-0/984637/4
- https://forum.arduino.cc/t/save-as-dialog-defaults-to-sketchbook-root-instead-of-previous-folder/1386329

However, those reports are from years ago, so they don't match with your observation that this is a regression introduced in 2.3.6 (or maybe 2.3.5).

per1234, 2025-06-06:
>If I save under another directory (not under /Arduino) then the "save as" it always takes me to /Arduino

OK, great. This is the essential information. I can now reproduce the fault once I produce the required conditions of saving the sketch to a location outside the sketchbook folder.

However, I am not able to reproduce this part of your report:

>until 2.3.4 did work correctly

The fault also occurs for me when using Arduino IDE 2.3.4.

Please perform this procedure and then report your results:

1. If a version of Arduino IDE other than 2.3.4 is running on your computer, select **File > Quit** from the Arduino IDE menus to close all Arduino IDE windows.
1. Start Arduino IDE 2.3.4.
1. Select **File > New Sketch** from the Arduino IDE menus.
   A new Arduino IDE window opens.
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch at `~/arduino_arduino-ide_2725-test/foo`.
1. Select **File > Save As...** from the Arduino IDE menus.

Now please comment here to tell us whether the Save As dialog opened in the expected location `~/arduino_arduino-ide_2725-test/`, or if it instead opened in the sketchbook folder.

> with v 2.3.4 I get the following interface, where I can also save "bookmarks" 

That is out of scope for this issue. You are welcome to open a separate issue to report this apparent regression related to "bookmarks", but please focus all discussion here exclusively on the "defaults to the Arduino default sketch directory (usually Documents/Arduino), regardless of where you last saved or opened a sketch" fault that is the subject of this issue.