arduino/arduino-ide #2726: プログラムファイルにカーソルを合わせた時の説明表示

miki-wmhne, 2025-05-01:
### Describe the request

プログラ

[image]

ムファイルを編集する時にカーソルを合わせると 説明のポップアップが時々邪魔をして編集が出来ない時が有ります

### Describe the current behavior

マウスカーソルを合わせた時に編集するプログラムに近付き過ぎているので表示位置の見直しをお願いします

### Arduino IDE version

Ver.2.3.6  + arduino-ide_nightly-20250429_Windows_64bit.zip

### Operating system

Windows

### Operating system version

Windows11 24H2

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

per1234, 2025-05-02:
Hi @miki-wmhne. Thanks for your suggestion. This "hover" feature comes from the framework upon which Arduino IDE is build, so it would not be feasible for the Arduino developers to modify its behavior. However, you can adjust the configuration according to your preferences via Arduino IDE's [advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings).

## Disable "Sticky" Behavior

I think that disabling the "**Editor › Hover: Sticky**" setting will solve the problem you are experiencing. I'll provide instruction you can follow to do that:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
   A menu will appear on the editor toolbar:
   [image]
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
   **ⓘ** You can scroll down through the list of commands to find it or type the name in the field.
   A "**Settings**" tab will open in the Arduino IDE main panel.
1. Type `editor.hover.sticky` in the "**Search Settings**" field of the "**Settings**" tab.
1. Uncheck the box under the "**Editor › Hover: Sticky**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

After this change to the settings, you will find that the "hover" disappears as soon as the mouse pointer is moved over it, so it will never interfere with your ability to select a position in the editor.

## Disable Hover

If you find the random appearances of the hover to be generally distracting, you can disable it:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
   A menu will appear on the editor toolbar.
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
   **ⓘ** You can scroll down through the list of commands to find it or type the name in the field.
   A "**Settings**" tab will open in the Arduino IDE main panel.
1. Type `editor.hover.enabled` in the "**Search Settings**" field of the "**Settings**" tab.
1. Uncheck the box under the "**Editor › Hover: Enabled**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

After disabling the feature, the "hover" will no longer appear when you hover the mouse pointer over code in the Arduino IDE editor. However, you can still trigger it manually when you do want to use it to get information about code:

1. Place the cursor on the code you want to get information about.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**K**</kbd> keyboard shortcut.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**I**</kbd> keyboard shortcut.
   **ⓘ** This is a [key sequence](https://github.com/eclipse-theia/theia/tree/master/packages/keymaps#key-sequences) (AKA "[chord](https://code.visualstudio.com/docs/getstarted/keybindings#:~:text=current%20context.-,Chords,-%28two%20separate%20keypress)"), where you must press two separate keyboard shortcuts in sequence.

You can also customize the shortcut according to your preference by selecting **File > Advanced > Keyboard Shortcuts** from the Arduino IDE menus and then adjusting the configuration for the "**Show or Focus Hover**" command.

## Other Settings

There are various other advanced settings which can be used to adjust the behavior of the hover according to your preference. I'll provide instructions you can follow to review and adjust these:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
   A menu will appear on the editor toolbar.
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
   **ⓘ** You can scroll down through the list of commands to find it or type the name in the field.
   A "**Settings**" tab will open in the Arduino IDE main panel.
1. Type `editor hover` in the "**Search Settings**" field of the "**Settings**" tab.
1. You will now see a list of settings, most of which are related to the behavior of the "hover" feature. Adjust these according to your preferences.
1. Close the **Preferences** tab by clicking its **X** icon.

closed 2025-05-02 by per1234

miki-wmhne, 2025-05-02:
>    1. Press the *Ctrl*+*Shift*+*P* keyboard shortcut (*Command*+*Shift*+
>    *P* for macOS users) to open the "Command Palette".
>    2. Select the "*Preferences: Open Settings (UI)*" command from the
>    3. Type editor.hover.sticky in the "*Search Settings*" field of the "
>    4. Uncheck the box under the "*Editor › Hover: Sticky*" setting.
>    5. Close the *Preferences* tab by clicking its *X* icon.
>    3. Type editor.hover.enabled in the "*Search Settings*" field of the "
>    4. Uncheck the box under the "*Editor › Hover: Enabled*" setting.
>    2. Press the *Ctrl*+*K* keyboard shortcut.
>    3. Press the *Ctrl*+*I* keyboard shortcut.
>    3. Type editor hover in the "*Search Settings*" field of the "
>    4. You will now see a list of settings, most of which are related to

It is welcome.
It became easy to use if it changes the establishment.