arduino/arduino-ide #2729: supress offline connection

cabiamdos, 2025-05-06:
### Describe the request

i would like to disable this yellow bar that tells me I am offline

[image]

it is really annoying

### Describe the current behavior

it appears at the bottom a yellow bar telling me I am offline

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

windows 10

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

per1234, 2025-05-08:
Hi @cabiamdos. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/2604.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

closed 2025-05-08 by per1234

cabiamdos, 2025-05-11:
how is that not planned. I belive you should rethink it

per1234, 2025-05-11:
@cabiamdos please carefully read the comment where I clearly explained the reason why I closed the issue.

In the future, please make a stronger effort to search the issue tracker before submitting issues to this repository so that we can both avoid the wasted time and effort that results from duplicate issues like this one.