arduino/arduino-ide #2740: Unable to start Arduino IDE

julien3108, 2025-06-13:
### Describe the problem

Hello,

I'm under Windows 11 with and without proxy,  when I run the Arduino 2.3.6 IDE and older 2.x.x versions, the application hangs like this:

[image]

Please find below the full log:

```
PS C:\Arduino\arduino-ide_2.3.6_Windows_64bit> & '.\Arduino IDE.exe'

PS C:\Arduino\arduino-ide_2.3.6_Windows_64bit> Arduino IDE 2.3.6
Checking for frontend application configuration customizations. Module path: C:\Arduino\arduino-ide_2.3.6_Windows_64bit\resources\app\lib\backend\electron-main.js, destination 'package.json': C:\Arduino\arduino-ide_2.3.6_Windows_64bit\resources\app\package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.6" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "1.2.0" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2025-04-09T11:26:55.498Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{},"showWindowEarly":true,"splashScreenOptions":{},"uriScheme":"arduino-ide"},"defaultLocale":"","validatePreferencesSchema":false,"reloadOnReconnect":true,"uriScheme":"theia","preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.6","cliVersion":"1.2.0","buildDate":"2025-04-09T11:26:55.498Z"}
Starting backend process. PID: 12220
Showing main window early
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///c%3A/Users/userName/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2025-06-13T09:07:13.122Z root INFO Backend u.initialize: 28.6 ms [Finished 0.579 s after backend start]
2025-06-13T09:07:13.123Z root INFO Backend Object.initialize: 27.1 ms [Finished 0.579 s after backend start]
2025-06-13T09:07:13.124Z root INFO Backend a.initialize: 1.7 ms [Finished 0.579 s after backend start]
2025-06-13T09:07:13.124Z root INFO Backend a.initialize: 1.4 ms [Finished 0.579 s after backend start]
2025-06-13T09:07:13.124Z root INFO Backend u.initialize: 1.8 ms [Finished 0.580 s after backend start]
2025-06-13T09:07:13.126Z root INFO Backend l.initialize: 28.7 ms [Finished 0.581 s after backend start]
2025-06-13T09:07:13.135Z root INFO configured all backend app contributions
2025-06-13T09:07:13.137Z root INFO Backend l.onStart: 2.3 ms [Finished 0.594 s after backend start]
2025-06-13T09:07:13.137Z root INFO Backend d.onStart: 0.8 ms [Finished 0.595 s after backend start]
2025-06-13T09:07:13.139Z root INFO Backend a.onStart: 0.5 ms [Finished 0.596 s after backend start]
2025-06-13T09:07:13.140Z config INFO >>> Initializing CLI configuration...
2025-06-13T09:07:13.140Z root INFO Backend x.onStart: 0.5 ms [Finished 0.597 s after backend start]
2025-06-13T09:07:13.140Z config INFO Loading CLI configuration from c:\Users\userName\.arduinoIDE\arduino-cli.yaml...
2025-06-13T09:07:13.141Z daemon INFO Starting daemon from C:\Arduino\arduino-ide_2.3.6_Windows_64bit\resources\app\lib\backend\resources\arduino-cli.exe...
2025-06-13T09:07:13.141Z root INFO Backend w.onStart: 1.0 ms [Finished 0.598 s after backend start]
2025-06-13T09:07:13.141Z discovery-log INFO start
2025-06-13T09:07:13.141Z discovery-log INFO start new deferred
2025-06-13T09:07:13.141Z root INFO Backend v.onStart: 0.6 ms [Finished 0.599 s after backend start]
2025-06-13T09:07:13.141Z root INFO Backend a.onStart: 0.2 ms [Finished 0.600 s after backend start]
2025-06-13T09:07:13.178Z root WARN The local plugin referenced by local-dir:/c%3A/Users/userName/.arduinoIDE/plugins does not exist.
2025-06-13T09:07:13.178Z root WARN The local plugin referenced by local-dir:/c%3A/Users/userName/.arduinoIDE/deployedPlugins does not exist.
2025-06-13T09:07:13.179Z root WARN The local plugin referenced by local-dir:C:\Users\userName\.arduinoIDE\plugins does not exist.
2025-06-13T09:07:13.184Z root INFO Theia app listening on http://127.0.0.1:62047.
2025-06-13T09:07:13.184Z root INFO Finished starting backend application: 43.2 ms [Finished 0.643 s after backend start]
2025-06-13T09:07:13.188Z config INFO Loaded CLI configuration: {"board_manager":{"additional_urls":["https://github.com/stm32duino/BoardManagerFiles/raw/dev/package_stmicroelectronics_index.json"]},"directories":{"builtin":{"libraries":"C:\\Users\\userName\\AppData\\Local\\Arduino15\\libraries"},"data":"c:\\Users\\userName\\AppData\\Local\\Arduino15","user":"c:\\Users\\userName\\OneDrive - STMicroelectronics\\Documents\\Arduino"},"locale":"en"}
2025-06-13T09:07:13.188Z config INFO 'directories.data' and 'directories.user' are set in the CLI configuration model.
2025-06-13T09:07:13.188Z config INFO Loaded the CLI configuration.
2025-06-13T09:07:13.189Z config INFO Mapped the CLI configuration: {"dataDirUri":"file:///c%3A/Users/userName/AppData/Local/Arduino15","sketchDirUri":"file:///c%3A/Users/userName/OneDrive%20-%20STMicroelectronics/Documents/Arduino","additionalUrls":["https://github.com/stm32duino/BoardManagerFiles/raw/dev/package_stmicroelectronics_index.json"],"network":"none","locale":"en"}
2025-06-13T09:07:13.189Z config INFO Validating the CLI configuration...
2025-06-13T09:07:13.189Z config INFO The CLI config is valid.
2025-06-13T09:07:13.189Z config INFO <<< Initialized the CLI configuration.
2025-06-13T09:07:13.300Z root INFO Resolve plugins list: 154.0 ms [Finished 0.756 s after backend start]
2025-06-13T09:07:13.302Z daemon INFO time="2025-06-13T11:07:13+02:00" level=info msg="arduino-cli version 1.2.0"
2025-06-13T09:07:13.303Z daemon INFO time="2025-06-13T11:07:13+02:00" level=info msg="Using config file: c:\\Users\\userName\\.arduinoIDE\\arduino-cli.yaml"
time="2025-06-13T11:07:13+02:00" level=info msg="Executing `arduino-cli daemon`"
2025-06-13T09:07:13.322Z daemon INFO Daemon is now listening on 127.0.0.1:62048
{"IP":"127.0.0.1","Port":"62048"}
2025-06-13T09:07:13.322Z daemon INFO Daemon is running.
2025-06-13T09:07:13.928Z root INFO Deploy plugins list: 785.5 ms [Finished 1.387 s after backend start]
2025-06-13T09:07:14.497Z root INFO creating connection for 1
2025-06-13T09:07:15.327Z root WARN Frontend a.configure took longer than the expected maximum 100 milliseconds: 153.0 ms [Finished 2.130 s after frontend start]
2025-06-13T09:07:15.351Z root ERROR Uncaught Exception:  Error: 14 UNAVAILABLE: No connection established. Last error: Failed to connect (2025-06-13T09:07:15.348Z)
2025-06-13T09:07:15.351Z root ERROR Error: 14 UNAVAILABLE: No connection established. Last error: Failed to connect (2025-06-13T09:07:15.348Z)
    at t.callErrorFromStatus (C:\Arduino\arduino-ide_2.3.6_Windows_64bit\resources\app\lib\backend\main.js:2:3089094)
    at Object.onReceiveStatus (C:\Arduino\arduino-ide_2.3.6_Windows_64bit\resources\app\lib\backend\main.js:2:6679669)
    at Object.onReceiveStatus (C:\Arduino\arduino-ide_2.3.6_Windows_64bit\resources\app\lib\backend\main.js:2:2572294)
[... 27 lines cut ...]
### To reproduce

Execute Arduino IDE.exe

### Expected behavior

Arduino IDE opened properly.

### Arduino IDE version

2.3.6

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

fpistm, 2025-06-24:
Hi @julien3108 ,
After some investigation on our side, it was an issue with proxy configuration. 
I think you can close this issue as not related to Arduino IDE.
Probably that  a timeout would help and an error message displayed if the Arduino IDE is unable to properly start.

julien3108, 2025-06-24:
Thanks Frederic for your time. The issue is resolved, it was an issue with proxy configuration. Julien