arduino/arduino-ide #2752: Multi-line comments causing build failure

nikthefix, 2025-07-06:
### Describe the problem

Recently I've found that simple alpha-numeric multi-line comments (no back slashes etc) are sometimes the cause of build failure in unpredictable ways. It appears to be associated with any white space lines between the comment block and code. Deleting blank lines always cures the problem but that is the only reproducible effect.

Examining the hex edit of an offending sketch reveals that in some cases (but not all) the UTF-8-BOM encoding header characters are the cause. If these characters are cut then the .ino builds just fine.

In other cases there are no such rogue characters yet the problem remains.

### To reproduce

Appears random in my experience so not reproducible by me. 

### Expected behavior

N/A

### Arduino IDE version

2.3.6

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

ritesh006, 2025-08-24:
**Confirmed root cause: UTF-8 BOM**

I was able to reproduce this consistently. It’s not random: the failures happen when the .ino file is saved as UTF-8 with BOM.

Environment

Arduino IDE: 2.3.6 (Windows 11)

Board: Arduino Uno — Arduino AVR Boards 1.8.6

FQBN: arduino:avr:uno

**Minimal sketch**
```
/* test */

int x = 42;

void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println(x);
  delay(1000);
}
```
**Steps to reproduce**

1. Save this sketch as UTF-8 with BOM (VS Code → Save with Encoding → UTF-8 with BOM).

2. Compile in Arduino IDE.

**Actual result**
```
error: stray '\357' in program
error: stray '\273' in program
error: stray '\277' in program
```
These are the 3 BOM bytes (EF BB BF).

**Control**
Re-save as UTF-8 (without BOM) → compiles fine.

**Notes**
Explains why it feels random: some editors add BOM silently.
The blank line after the comment makes the bug surface consistently.

**Proposed fix**
When reading .ino files, strip the BOM before preprocessing. For example, skip the first 3 bytes if they are EF BB BF.

**Workaround**
Save .ino as UTF-8 (no BOM).

**Artifacts**
**PowerShell check of first 3 bytes:**
```
# Show first 3 bytes of the .ino file in hex
$b = Get-Content .\BomBug\BomBug.ino -AsByteStream -TotalCount 3
$b | ForEach-Object { '{0:X2}' -f $_ }
```

**Output:**
```
EF
BB
BF
```

ritesh006, 2025-08-24:
“I’ve updated the patch, and after merging, this issue is unlikely to occur again.

nikthefix, 2025-08-24:
@ritesh006 Nice work!