arduino/arduino-ide #2753: Capital letters doesnt work in dev board search and its acting weird.

MYCHMA, 2025-07-07:
### Describe the problem

You cant write Capital letters in searchbox, and if I hit the option that I want the search box content gets deleted,the options thanks god applies.

### To reproduce

You cant write Capital letters in searchbox

### Expected behavior

You can write Capital letters in searchbox,and the query stays until you close it or you delete it and you can choose multiple times from the list of found results

### Arduino IDE version

2.3.6

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

ritesh006, 2025-08-28:
Oh, you mean here you can’t type capital letters? Is that really a big issue? Because when I type in lowercase, the board is still visible. Is there some issue I might be missing?

[image]

MYCHMA, 2025-08-28:
Read all of it before you make conclusion:  You cant type caps. And it CANT remember the text inside.
So when I want to change board I need to type it aggain and aggain.

per1234, 2025-10-02:
Hi @MYCHMA. Thanks for taking the time to submit an issue.

The conversion of the search query to lowercase is done intentionally. The search is case insensitive so this has no functional effect.

Since that is the primary subject of this issue, I will closed it.

As for the secondary subject of remembering the search query, you are welcome to submit a dedicated feature request for that. In the future, please be careful to restrict the issues you submit to a single subject. Mashing multiple distinct subject matters into a single issue makes it impossible for us to effectively track the report.

Please also make a stronger effort in writing any issues you submit to our issue trackers in the future. Your issue was quite low quality and I had to waste some time in trying to understand it and then rewriting it to meet our quality standards.

closed 2025-10-02 by per1234