arduino/arduino-ide #2759: Segmentation fault when launching Arduino IDE 2.3.6 on Arch Linux

abod8639, 2025-07-17:
### Describe the problem

Launching the Arduino IDE (v2.3.6) on Arch Linux causes an immediate crash with a segmentation fault. The issue occurs right after frontend configuration is initialized. The program exits before rendering any user interface.

### To reproduce

Arduino IDE 2.3.6
Checking for frontend application configuration customizations. Module path: /opt/arduino-ide/resources/app/lib/backend/electron-main.js, destination 'package.json': /opt/arduino-ide/resources/app/package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.6" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "1.2.0" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2025-04-09T11:24:04.835Z" (type of string)
Frontend application configuration after modifications: {...}
zsh: segmentation fault (core dumped)  arduino-ide

### Expected behavior

The IDE should initialize and display the GUI without crashing.

### Arduino IDE version

2.3.6

### Operating system

Linux

### Operating system version

      Operating System: Arch Linux (fully up-to-date)      Kernel: Linux 6.x (latest available)      Shell: zsh      Window System: Wayland (Hyprland)      Arduino IDE Version: 2.3.6      Installation Method: pacman or yay      Electron Version (bundled): as shipped with IDE      Desktop Environment / WM: Hyprland (Wayland compositor)

### Additional context

    The crash occurs consistently on every launch.

    It appears to be related to Electron or frontend initialization in a Wayland environment.

    Other Electron apps may work correctly, but this issue seems specific to this build of Arduino IDE.

Is there a known compatibility issue with Wayland (Hyprland in particular)?

Are there any environment flags (e.g. forcing X11) that can work around this issue?

Would it help to downgrade or rebuild Electron dependencies manually?

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

abod8639, 2025-07-19:
Thank you for your response.

I tested both versions 2.3.6 and 2.3.5, and unfortunately the issue still occurs in both — even when downloaded directly from the official Arduino website.

However, I tried versions 2.3.4 and 2.3.3, and both worked normally without any issues, regardless of whether I installed them from the official source or from the Arch Linux repository.

Let me know if you'd like me to provide additional logs or test specific scenarios.

kjkent, 2025-08-22:
@abod8639 you might be experiencing https://github.com/arduino/arduino-ide/issues/2107, potentially moreso if you have `CHROMIUM_FLAGS='--enable-features=...'` set in your environment for, e.g. NVIDIA compatibility.

You can try unsetting the above env var, and if that doesn't work, try running with the following:

```Shell
env -u WAYLAND_DISPLAY XDG_SESSION_TYPE=x11 GBM_BACKEND=dri arduino-ide
```

fisika1729, 2025-09-03:
> Thank you for your response.
>
> I tested both versions 2.3.6 and 2.3.5, and unfortunately the issue still occurs in both — even when downloaded directly from the official Arduino website.
>
> However, I [...]

while the 2.3.4 build works now, seg fault happened for that as well when it's release, because of fuse and libfuse2 dependency.

GZGavinZhao, 2025-09-21:
For me `arduino-ide --enable-features=UseOzonePlatform` is still using XWayland. If I unset `DISPLAY` then it crashes with `[152891:0920/232324.354098:ERROR:ozone_platform_x11.cc(244)] Missing X server or $DISPLAY`.

GZGavinZhao, 2025-09-21:
I did some investigation and it seems like the root cause is from us using an older version of Electron (30). I suspect if we upgrade to newer Electron versions it might just work. Are there any plans to bump the Electron version?

KorewaLidesu, 2026-03-14:
> It’s not working on Niri, but it works perfectly fine on Sway.

It does work on Niri, but you need to run it on XWayland (append `--enable-features=UseOzonePlatform --ozone-platform=x11` to `~/.config/arduino-flags.conf`)
[image]

cleudopaiva, 2026-04-19:
For those experiencing scaling issues in the user interface, such as in the menu bar, you can try forcing the scaling setting with `--force-device-scale-factor=1`.

In the end this is my `~/.config/arduino-flags.conf` file: 
`--enable-features=UseOzonePlatform --ozone-platform=x11 --force-device-scale-factor=1`

It's happening on the Omarchy + Hyprland

Z00Li, 2026-08-29:
Confirmed on Arduino IDE 2.3.10, Arch Linux with the Hyprland compositor (Wayland). Same crash, same fix.

## Environment
- Arduino IDE: 2.3.10 (packaged as `arduino-ide-bin`; bundles **Electron 30.1.2 / Chrome 124 / Node 20.14**)
- OS: Arch Linux (glibc 2.44), fully up to date
- Compositor: Hyprland (Wayland). The IDE refuses to run natively under Wayland and segfaults on a consistent basis.

## Crash signature
The main process dies with `SIGSEGV` (signal 11) moments after printing the frontend config:

```
Arduino IDE 2.3.10
Checking for frontend application configuration customizations. ...
Setting 'theia.frontend.config.appVersion' ...
...
Frontend application configuration after modifications: {...}
<segfault>   (core dumped)
```

`coredumpctl` details:
- Signal: 11 (SEGV), `si_code: SI_TKILL` (the Chromium/Electron main process aborts itself deliberately, i.e. an internal fatal path, rather than a plain null dereference)
- Crashing thread is the main process in V8 JIT code (unsymbolizable `??` frames); all worker threads are idle/blocked
- Not OOM (plenty of free RAM, no OOM-killer entries)
- Reproduces on every launch; identical coredump stack across runs

This is the **bundle's stale Electron 30 native Wayland backend** being incompatible with Hyprland. The Electron upgrade is tracked in #2424.

## Solution (verified working)
Force the X11 (XWayland) Ozone backend. On Arch the launcher script already reads `~/.config/arduino-flags.conf`, so the fix is simply:

`~/.config/arduino-flags.conf`:
```
--enable-features=UseOzonePlatform
--ozone-platform=x11
```

Equivalently, launch with:
```
arduino-ide --enable-features=UseOzonePlatform --ozone-platform=x11
```

Verification on this system: after adding these flags the IDE boots a full, healthy process tree (gpu-process, renderer, network service, backend `main.js`, `arduino-cli` daemon) under `--ozone-platform=x11` and stays running indefinitely — no coredump. Prior to the flags it always crashed within ~1–2 s of launch.

Note: this runs the IDE through XWayland, so it is a workaround rather than native Wayland support. The proper fix is upgrading the bundled Electron, per #2424.

CezarVeleda, 2026-09-29:
Still reproducible on 2.3.10 (AUR arduino-ide-bin 2.3.10-1, bundles Electron 30.1.2 / Chrome 124.0.6367.243), and I was able to narrow down the trigger.

Environment
- Arch Linux (Omarchy), kernel 7.2.5
- Hyprland 0.56.2
- NVIDIA RTX 3090, proprietary driver 610.57.04
- mesa 26.2.2, wayland 1.26.0

Symptom
Same as the original report: SIGSEGV about 1 second after launch, right after the Frontend application configuration after modifications log line, before any window appears. It happens on every launch, including the very first one with a fresh profile. systemd-coredump shows the faulting frame inside the main arduino-ide binary (unsymbolized, no debug symbols available).

Trigger
The session exports ELECTRON_OZONE_PLATFORM_HINT=wayland (Omarchy sets this by default, and it's common on Hyprland setups). With it, Electron selects the native Wayland Ozone backend and the IDE crashes. Forcing X11/XWayland makes the crash go away:

# crashes
arduino-ide

# works: IDE starts normally, first-run setup completes
env -u ELECTRON_OZONE_PLATFORM_HINT arduino-ide --ozone-platform=x11

Workaround
Override the desktop entry in ~/.local/share/applications/arduino-ide-v2.desktop with:

Exec=env -u ELECTRON_OZONE_PLATFORM_HINT arduino-ide --ozone-platform=x11 %U

Since Electron 30 is quite old at this point and native Wayland support has improved a lot in later Electron releases, upgrading Electron may resolve this. Otherwise, defaulting to --ozone-platform=x11 on Linux until native Wayland works would stop the IDE from crashing on first launch for Hyprland users.