arduino/arduino-ide #2760: Arduino IDE 2.3.6 Still Faulty on Auto Update on MacOS

DannyV1, 2025-07-23:
### Describe the problem

It works for a bit then hangs on a big lib: (See capture below).
Mac OS 13.7.6  (see screen grab at end).

CAPTURE:
_Downloading Adafruit NeoPixel@1.15.1
Adafruit NeoPixel@1.15.1
Installing Adafruit NeoPixel@1.15.1
Replacing Adafruit NeoPixel@1.12.4 with Adafruit NeoPixel@1.15.1
Installed Adafruit NeoPixel@1.15.1
Downloading Adafruit PN532@1.3.4
Adafruit PN532@1.3.4
Installing Adafruit PN532@1.3.4
Replacing Adafruit PN532@1.3.3 with Adafruit PN532@1.3.4
Installed Adafruit PN532@1.3.4
Downloading Adafruit SPIFlash@5.1.1
Adafruit SPIFlash@5.1.1
Installing Adafruit SPIFlash@5.1.1
Replacing Adafruit SPIFlash@5.0.1 with Adafruit SPIFlash@5.1.1
Installed Adafruit SPIFlash@5.1.1
Downloading Adafruit SSD1306@2.5.15
Adafruit SSD1306@2.5.15
Installing Adafruit SSD1306@2.5.15
Replacing Adafruit SSD1306@2.5.13 with Adafruit SSD1306@2.5.15
Installed Adafruit SSD1306@2.5.15
(Hangs)
Next opening of IDE shows lib Adafruit SSD1306 installed....
Mac About:
[image]

### To reproduce

On a MAC, just run Auto Update and wait. No cards attatched, many cards need updates also.

### Expected behavior

IDE Hangs on large lib

### Arduino IDE version

IDE 2.3.6

### Operating system

macOS

### Operating system version

Mac OS 13.7.6 

### Issue checklist

- [x] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My report contains all necessary details

DannyV1, 2025-07-23:
I Think the libraries that it hung up on were the ones that had the subscript _library on them (not sure about the case of the 'L' ), as when I checked that dir, the libs that it hung up on still had that subscript on it, waiting for me to remove it.
I can't be sure it was 100% the case though.

closed 2025-07-23 by DannyV1

Emilurenius, 2025-09-24:
I have had the same issue on Fedora 42 with the flatpak installation of the IDE version 2.3.6.
If I try to update more than one library at a time, it just hangs and finishes none of them. Force closing the program and then installing them one by one worked for me

pavelpetrovic, 2026-03-01:
It still happens in the latest Arduino 2 IDE.