arduino/arduino-ide #2775: Better serial monitoring/interaction

bixbyru, 2025-08-24:
### Describe the request

Hey all - 

The Arduino serial monitor and IDE code uploader share a serial line such that each is disconnected from the serial port while the other needs it. 

Since this routing mechanism is already in place, it seems that it would not be that difficult to provide a means to route traffic from the USB serial connexion to an external virtual port which would be accessible via an xrerm, PuTTY or what have you; this would improve the usability of serial monitoring and interaction considerably. 

As a for instance, ANSI escape sequences make complex display of system status much more straightforward but are so much gobbledygook on the built-in serial monitor. 

Currently this is accomplished by killing one program and starting the other, switching back and forth as required, but this is suboptimal to say the least. 

Thanks...? 

### Describe the current behavior

Serial monitor works, I guess, but for anything but the most basic interaction it's as useless as a rubber anvil. 

### Arduino IDE version

All versions 

### Operating system

N/A

### Operating system version

Modern Linux, Windows 10/11 and OS/X 

### Additional context

Nuthin' else to say, really. 

### Issue checklist

- [x] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [x] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [x] My request contains all necessary details

ritesh006, 2025-09-08:
@bixbyru I hope your problem is solved

johonline1, 2025-12-19:
I've been using a second USB port connected to TX and GND, and I've been using it with a terminal.

I find the virtual port- solution very advantageous.